Klipper3d/klipper #5721: Implicit Z move triggers home axis first error.

rewolff, 2022-08-21:
I thought I'd mention this on the discord server, but I can't get along with your discord. It asks me to claim my account and when I enter my details (emai/password) it says it's already claimed. No way to "switch accounts" to log in with my existing account. etc etc. I give up. 

---------------------------------------------------------------
I upgraded my machine yesterday to have an automatic bed leveling probe. 

I just finished printing something, and I'm not where the printer is. But I can see the result on the camera. To maneuver the print to get a better look, I have to "home XY" first because the motors have been turned off. Then as long as I don't touch Z, I should be able to move the bed around in XY, rigth? Wrong: A move in Y will implicitly also command a Z move and therefore error out on "you must home axis first". I can't home Z now because the object I just printed is in the way.

jakep82, 2022-08-21:
Please post help requests on Discourse if you're unable to use Discord, and always post a log.  Without a log all I can do is guess the reason you can't move in X or Y is because you have a mesh loaded.  You need to clear the mesh.

rewolff, 2022-08-21:
Yes, I have a mesh loaded. I noticed the Z axis doing small movements when I checked out the beginning of this print. 

Your guess is correct. I "have a mesh loaded". 

1) How would i go about clearing the mesh ?

2) I find this "undesirable behaviour". I'm just maneuvring the head in XY. if the Z axis hasn't been homed, then maybe tolerate the small movements due to the mesh, as opposed to insisting on the home operation.

closed 2022-08-23 by github-actions[bot]