Klipper3d/klipper #5723: ADC out of range - improve logging

bhelm, 2022-08-21:
While debugging a 'ADC out of range' i noticed that the status line just before the error eventually does not capture the out-of-range value (maybe 10% chance for the status line just before the 'ADC out of range' to not show the error value).

People on discord told me that its already hard to tell which ADC failed, so having an status line in the timeframe the error is triggered could make things easier.

Looking at the code in src/adccmds.c around line 49 makes me believe that the ADC triggers an error after 0,9 seconds of bad values while the status line appears every 1 second. first error on 0s, 2nd on 0.3s, 3rd on 0.6s and last one on 0.9s triggers the shutdown.

My suggestion is to either increase the RANGE_CHECK_COUNT to a sane value so it guarantees a status line or (even better) push a final status line when a shutdown is triggered.

Additionally, it would be nice if that status line would include ALL sensor values. The value of temperature_fan psu is missing.

My actual problem is that i got that error mid-print but after restart the printer works fine (i suspect a bed cable or thermistor). I want to know which sensor triggered the error so i can just replace the thermistor and avoid the problem to happen again mid print. Its on line 4451, if it can be told from the log, an hint would be appreciated.
[klippy (3).log](https://github.com/Klipper3d/klipper/files/9389876/klippy.3.log)

I have 3 sensors on the system where two are on the status log line, which means that if it is guaranteed that the error would show up there, i could tell its the 3rd sensor not in the status line.

I did tests on whenever the bad temperature is reported after i disconnected the bed's thermistor, see these lines:
11533: disconnected bed, error temp <-50° visible in stats
12655: disconnected bed, error temp <-50° visible in stats
13808: disconnected bed, bed shows correct room temperature
15708: test with RANGE_CHECK_COUNT =20

Sineos, 2022-08-22:
As already noted when you opened this topic: GitHub shall only be used for development purposes. Please be so kind to open such support or feature requests topics on [Klipper Discourse](https://klipper.discourse.group/)

bhelm, 2022-08-22:
Ok im sorry.

closed 2022-08-22 by bhelm