Klipper3d/klipper #5727: Raspberry pi Pico(RP2040) unable to enumerate usb device

0LYZERO0, 2022-08-24:
Board doesn't enumerate properly after bootup or a USB disconnect/reconnect. 

After USB disconnect/reconnect
[  486.895871] usb 1-1.2: USB disconnect, device number 15
[  488.317369] usb 1-1.2: new full-speed USB device number 16 using dwc_otg
[  488.527389] usb 1-1.2: device descriptor read/64, error -32
[  488.847366] usb 1-1.2: device descriptor read/64, error -32
[  489.177381] usb 1-1.2: new full-speed USB device number 17 using dwc_otg
[  489.377368] usb 1-1.2: device descriptor read/64, error -32
[  489.697409] usb 1-1.2: device descriptor read/64, error -32
[  489.817473] usb 1-1-port2: attempt power cycle
[  490.577355] usb 1-1.2: new full-speed USB device number 18 using dwc_otg
[  491.017387] usb 1-1.2: device not accepting address 18, error -32
[  491.217424] usb 1-1.2: new full-speed USB device number 19 using dwc_otg
[  491.657397] usb 1-1.2: device not accepting address 19, error -32
[  491.657547] usb 1-1-port2: unable to enumerate USB device

After pressing reset button

[  494.717429] usb 1-1.2: new full-speed USB device number 20 using dwc_otg
[  494.951673] usb 1-1.2: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
[  494.951699] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  494.951714] usb 1-1.2: Product: rp2040
[  494.951729] usb 1-1.2: Manufacturer: Klipper
[  494.951744] usb 1-1.2: SerialNumber: 455035712814D208
[  494.953083] cdc_acm 1-1.2:1.0: ttyACM1: USB ACM device

Sineos, 2022-08-24:
As already noted when you opened this topic: GitHub shall only be used for development purposes. Please be so kind to open such support or feature request topics on [Klipper Discourse](https://klipper.discourse.group/)

0LYZERO0, 2022-08-24:
This should be a bug. I tested at least four different boards with RP2040，they have the same problem

closed 2022-08-26 by github-actions[bot]