Klipper3d/klipper #5730: Duplication and mirror mode

novarlynx, 2022-08-24:
Sorry to be a bother but every time I've asked this in the discord channel I've been completely ignored or it got buried among other conversation. 

I have an IDEX printer and i've been wanting duplication & mirroring mode for a very long time. Seems there was a pull request a long time ago that had discussion going then the GitHub bot closed it and it got forgotten about.

https://github.com/Klipper3d/klipper/pull/4464

Sorry if this gets me in trouble, but please at least look. Thanks

novarlynx, 2022-08-24:
I spent an afternoon trying to get the code in the PR to work but I was unsuccessful. I have a certesian IDEX machine. I can find two examples of it working on a cartesian IDEX printer so I must be doing something wrong. 

If this gets reopened and worked on again I would be willing to test on my printer.

Sineos, 2022-08-25:
If there is no answer in the Discord / Discourse then most likely nobody is able to contribute currently. Makes no sense to cross post

closed 2022-08-27 by github-actions[bot]