Klipper3d/klipper #5740: Issue with Neopixels

Guilouz, 2022-08-27:
How many LEDs does Klipper support?

Because I have strange issue, I have a I have a 60 led ring with this config :

[neopixel RGB1]
pin: PD0
chain_count: 60
color_order: GRB
initial_RED: 1
initial_GREEN: 1
initial_BLUE: 1

At startup all my 60 led are well lit white. No problem.

But when I try to change colors with this macro, all my leds are not changed.
When I try to light off only part light off.

If I set as example chain_count: 30 all is working as expected.

This is my macros : 

[gcode_macro NEOPIXEL_ON]
description: Allumer les LEDs
gcode:
  SET_LED LED=RGB1 RED=1 GREEN=1 BLUE=1

[gcode_macro NEOPIXEL_OFF]
description: Eteindre les LEDs
gcode:
  SET_LED LED=RGB1 RED=0 GREEN=0 BLUE=0

      
[gcode_macro NEOPIXEL_BLUE]
description: Allumer les LEDs en bleu
gcode:
  SET_LED LED=RGB1 RED=0 GREEN=0 BLUE=1

[gcode_macro NEOPIXEL_RED]
description: Allumer les LEDs en rouge
gcode:
  SET_LED LED=RGB1 RED=1 GREEN=0 BLUE=0

[gcode_macro NEOPIXEL_GREEN]
description: Allumer les LEDs en vert
gcode:
  SET_LED LED=RGB1 RED=0 GREEN=1 BLUE=0

JamesH1978, 2022-08-28:
Please use the klipper discord or the discourse for this. We do not use the GitHub issues for general support, we only use it for actual bugs.

Guilouz, 2022-08-28:
@JamesH1978 Already posted on Discord without without answers...

closed 2022-08-30 by github-actions[bot]