Klipper3d/klipper #5750: BL-Touch offset drift with speed

mgth, 2022-09-03:
Trying to debug inconsistent bl-touch result, I got this interesting result, it seams that there is about 8ms "missing" from the returned value, while repeatability is ok at any speed, the distance measured reduce by about the distance done in 8ms at any speed.

[image]
[klippy(1).zip](https://github.com/Klipper3d/klipper/files/9481596/klippy.1.zip)

Ramalama2, 2022-10-11:
> Trying to debug inconsistent bl-touch result, I got this interesting result, it seams that there is about 8ms "missing" from the returned value, while repeatability is ok at any speed, the distance [...]

Do you have a script for that?
That's something im seeing with Microswitches either, but i was never able to convert that in a such beautiful graph.

Especially interesting is first probe either.
We know that first probe always sucks, but im wondering if that has todo with that either because the mcu somewhat sleeps or the code simply adds the delay on first probe.

Sineos, 2022-10-11:
Surely an interesting and worthwhile discussion. And yes, the aberrant first probe is a pain but please be so kind to move this discussion to https://klipper.discourse.group

closed 2022-11-24 by github-actions[bot]