Klipper3d/klipper #5753: [mcu host] Can not update MCU 'mcu' config as it is shutdown after system reset/start

Bauschlumpf, 2022-09-03:
Hello,

If the MCU host is activated in the printer.cfg the error message "Can not update MCU 'mcu' config as it is shutdown" appears after a system start or restart.

# Raspberry Pi as MCU
[mcu host]
serial: /tmp/klipper_host_mcu

The mcu host is only integrated and has no further function (integration test).
FIRMWARE_RESTART fixes the problem temporarily until the next reboot.
After deactivating the mcu host klipper starts without problems.

[klippy.log](https://github.com/Klipper3d/klipper/files/9482627/klippy.log)

Sineos, 2022-09-03:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [klipper.discourse.group](https://klipper.discourse.group/)

Bauschlumpf, 2022-09-04:
SORRY, I moved this to klipper.discourse.group General Discussion
Thank you

closed 2022-09-04 by Bauschlumpf