Klipper3d/klipper #5765: delta z_offset_apply_endstop unknown command

lnquang, 2022-09-09:
unable to update baby steps since command is broken. only for delta printer.

lnquang, 2022-09-09:
[image]

Sineos, 2022-09-09:
And as already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [klipper.discourse.group](https://klipper.discourse.group/)

lnquang, 2022-09-09:
this is a bug, not for help, please respond accordingly.

Sineos, 2022-09-09:
If you can pinpoint the lines of code that are causing the bug, then we can continue here. Otherwise please be so kind to follow the directions above and also provide your klippy.log there.
This ticket will not be followed up any more.

lnquang, 2022-09-10:
i am no developer, my klipper control 4 printers, 2 delta printers and 2 others, those delta printers did not recognize command above.

jakep82, 2022-09-10:
There is no Z endstop on a delta, so it would be expected that this command doesn't work.

closed 2022-09-12 by github-actions[bot]

lnquang, 2022-09-13:
right, you need somehow to saving babystepping, right now there is no way to save these adjustment.

ksdfpr, 2022-10-03:
Hi, I have the same problem. If the save button doesnt work how can we save a new z offset (in easy way) permanently?
Now, every time I turn on the printer I have to remember to change z offset manually.
And why does it show this button if it doesn’t work on delta printers? Maybe this works??? Z_OFFSET_APPLY_PROBE