Klipper3d/klipper #5772: Unable write i2c device

shogowarrior, 2022-09-13:
I recently connected 128x64 SSD1306 to RPI Zero W 2 as a secondary MCU and get the following error:
`Unable write i2c device`  

```log
Receive: 99 1512.122333 1512.121805 12: seq: 19, shutdown clock=2896182194 static_string_id=Command request
MCU 'host' shutdown: Unable write i2c device
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

MCU 'host' shutdown: Unable write i2c device
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Stats 1513.0: gcodein=0  mcu: mcu_awake=0.156 mcu_task_avg=0.000150 mcu_task_stddev=0.000295 bytes_write=90396 bytes_read=37563 bytes_retransmit=9 bytes_invalid=0 send_seq=3035 receive_seq=3035 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=11 stalled_bytes=0 freq=63998910 host: mcu_awake=0.087 mcu_task_avg=0.000992 mcu_task_stddev=0.001647 bytes_write=9519 bytes_read=6269 bytes_retransmit=0 bytes_invalid=0 send_seq=364 receive_seq=364 retransmit_seq=0 srtt=0.002 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999913 adj=50000726  heater_bed: target=60 temp=50.7 pwm=1.000 rpi: temp=42.9 mcu: temp=36.2 sysload=0.47 cputime=55.336 memavail=241228 print_time=47.283 buffer_time=1.140 print_stall=0 extruder: target=220 temp=202.5 pwm=0.835
Stats 1514.0: gcodein=0  mcu: mcu_awake=0.156 mcu_task_avg=0.000150 mcu_task_stddev=0.000295 bytes_write=90639 bytes_read=37930 bytes_retransmit=9 bytes_invalid=0 send_seq=3044 receive_seq=3044 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=63998914 host: mcu_awake=0.087 mcu_task_avg=0.000992 mcu_task_stddev=0.001647 bytes_write=9755 bytes_read=6441 bytes_retransmit=0 bytes_invalid=0 send_seq=371 receive_seq=371 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999914 adj=50000685  heater_bed: target=60 temp=51.0 pwm=1.000 rpi: temp=42.9 mcu: temp=36.0 sysload=0.43 cputime=55.411 memavail=241152 print_time=47.284 buffer_time=0.140 print_stall=0 extruder: target=220 temp=203.9 pwm=0.834
```

shogowarrior, 2022-09-13:
I also see`Neopixel update did not succeed` with the leds in weird state

JamesH1978, 2022-09-14:
As the automation says, please post this on the discourse or discord. so it can be triaged. GH issues are only used for known bugs.

Thanks
James

shogowarrior, 2022-09-15:
I posted it in discord also in #configuration

closed 2022-09-18 by github-actions[bot]