Klipper3d/klipper #5773: SPI3 not correctly assigned on STM32H7XX MCU based boards

jaysuk, 2022-09-15:
the issue is this line here https://github.com/Klipper3d/klipper/blob/master/src/stm32/stm32h7_spi.c#L70
it assigns all 3 SPI3 pins to function 6
but on the H742x/H743x at least that would configure PB4 and PB3 for SPI3 but PB5 would end up talking to the I2C4 device! To get PB5 to talk to SPI you need to set it's alternate function number to 7 not 6, which it appears that the current code doesn't let you do.

Look at page 89 of datasheet: https://www.st.com/resource/en/datasheet/stm32h743vi.pdf 
This issue specifically affects the fly-super8pro board, that uses hardware SPI 3 for TMC driver control

@adelyser I believe you were the original author of this code.

adelyser, 2022-09-25:
I did author this code, and you are correct, PB5 needs a different function number. I didn't have anything that tested that particular SPI3 connection, so it was missed. I can implement a change to fix that, but I'll need some help testing that it works since I don't have a device to test with.

jaysuk, 2022-09-25:
Of course. Happy to test whenever you've got something.

adelyser, 2022-09-25:
> Of course. Happy to test whenever you've got something.
> […](#)

You can test with the PR linked to this issue

adelyser, 2022-09-30:
> Of course. Happy to test whenever you've got something.
> […](#)

So it does seem you could just use SPI1a for those pins, correct?

KevinOConnor, 2022-11-01:
This should be fixed (by removing spi3) in commit e4c8a247.

-Kevin

closed 2022-11-09 by github-actions[bot]