Klipper3d/klipper #5774: Two C Stepper issue on Delta

macgyver1307, 2022-09-15:
Good day, 

I built a 48v AWD  Delta. I'm running an SKR pico for my main board ( heaters, therms, and fans) and 2 Supernova boards from Annex engineering at 48v for all steppers with Bigtree 5160 pros( annex products are fantastic). 
I'm having issues with Klipper and running 2 C steppers. I have no problem with my config running Stepper A, A1 and Stepper B, B1 but when I add Stepper C, C1 only the C Stepper moves not the C1 with it. The config for the steppers and driver are the same except  for the name change ( a,a1,b,b1,c,c1, extruder).
Testing this, I've tried different steppers and stepper drivers ( fysec to Bigtree). I've tried running an octopus Pro with the same results ( even after changing stepper ports and drivers ). Force moves and Buzzz commands do work but with no power behind them( stepper can feel like a command is working but very lightly not enough to move but enough to feel the stepper flutter). my last attempt was to verify its not a hardware issue but moving steeper names around in my config. A&A1 to B&B1 - B&B1 to C &C1 -  C&C1 to A&A1.  Each time  I had the same issue with C1  not moving correctly no matter the Tower I used. 

Attached is my config example 
[image]

Thank you for your help  I do not know what else I can look for.

Sineos, 2022-09-15:
As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [klipper.discourse.group](https://klipper.discourse.group/)

Also attache your klippy.log there

macgyver1307, 2022-09-15:
> As already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: [klipper.discourse.group](https://klipper.discourse.group/)
> 
> Also attache your klippy.log there

I have done this with no help on discord and I believe this is an issue or a request to enable multiple C steppers in kipper.   I will try again on discord.

macgyver1307, 2022-09-15:
Moved to Klipper discourse per instruction https://klipper.discourse.group/t/multi-motor-delta-c-stepper-issue/4144

closed 2022-09-15 by macgyver1307