Klipper3d/klipper #5783: Timer too close during resonance calibration with CAN bridge

abukharov, 2022-09-22:
Hi all,

I have a CoreXY on Octopus Pro F429 with HV 5160Pro drivers on X and Y. My microsteps are set to 16 with interpolation. I use an RPi 3B+ for host and a BTT EBB36 board connected via CAN, which has the accelerometer. CAN speed is set to 500000. Both Octopus and EBB36 are flashed with CanBoot bootloader and Klipper v0.10.0-584-g7527e57e

When I run resonance calibration on X I get a Timer Too Close when the frequency runs slightly over 100Hz. If I limit the resonance tester to 90Hz it finishes successfully. I've checked CPU usage, memory usage, voltage, temperature and the sdcard for errors, everything seems to be fine.

I'm actually puzzled, everything seems to be fine, but I consistently keep getting this error. Please let me know if you need any more info, I'm happy to test stuff and so on.

[klippy.log](https://github.com/Klipper3d/klipper/files/9629416/klippy.log)

ReXT3D, 2022-09-23:
As @jakep82 pointed out this has been discussed at length on Discourse and also on Discord, which is where all non-development related support questions and requests should go.  FWIW I will add the following here for reference.

Running just `MEASURE_AXES_NOISE` results in over 30% CAN bus load at 1000000:

```
pi@creality:~ $ canbusload can0@1000000
 can0@1000000    27    3345   1488   0%
 can0@1000000    23    2835   1256   0%
 can0@1000000   857  109595  49968  10%
 can0@1000000  2526  328260 151464  32%
 can0@1000000  2512  326370 150568  32%
 can0@1000000    77    9795   4448   0%
 can0@1000000    27    3345   1488   0%
```

abukharov, 2022-09-23:
Yeah, it makes sense, I'll give it a go and see. I've also found a workaround: limit the frequency to 90hz, my resonances are below 90 anyways.

I searched through discourse and didn't find anything related (maybe I just can't search) . There is a remotely related question in discord, but with no answer.

Thank you so much for your help.

closed 2022-09-23 by abukharov