Klipper3d/klipper #5793: SPI1 on BTT Manta M8P (STM32G0) issue

FranzPVal, 2022-09-26:
If I connect a ADXL345 to the SPI1 connector (PA5,6,7 + PB15) the firmware will not boot. No serial device pops up in the Linux host system.

I use this board:
https://www.aliexpress.com/item/1005001432536778.html
It has 4.7K pullups installed on SCK, SDA and CS installed and a pulldown on SDO. I'm not sure if this can cause a problem. I guess PA6,6,7 of the STM32G0B1 don't have special boot states.

controller board: BTT Manta M8P equipped with a CB1 Linux compute module replacement.

Is here a condition that prevent the firmware from boot?

Sineos, 2022-09-27:
And as already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

FranzPVal, 2022-09-27:
I don't understand what's going on here?! @Sineos are you a bot too?

I'm a developer. I'm working on STM32 targets and on embedded Linux os myself. I posted an issue here. There is clearly a bug I described and all I got is a text block?
Ok. I' know a bit about the STM32G0 SPI, I coded a TFT driver for it but I have no time to debug the klipper mcu firmware. All I tried is to describe the bug (hanging / not starting if a working ADXL board is connected).

Sineos, 2022-09-27:
The general philosophy here is:
* Either you can point to a faulty piece of code OR
* You **suspect** that something is wrong and then it is triaged via the discord, e.g. the developer section there

If you already know it is a bug then you should be able to pinpoint it more precisely. Otherwise you will depend on some core developer willing / able to find the time to dig into it.
When looking at the open PRs right now, this is unlikely to happen anytime soon (unless a trivial fix)

closed 2022-09-29 by github-actions[bot]