Klipper3d/klipper #5795: Shutdown behavior - BTT Manta M8P + CB1

FranzPVal, 2022-09-27:
The MCU firmware will not shutdown / go into the reset state if the Linux host was shutdown.

To replicate:
Turn on steppers, part cooling fan then call "sudo shutdown -h now" from a console or use the Mailsail UI "Host Control" -> "Shutdown" menu item.
This results in a shutdown of the CB1 Linux host board, but the MCU (STM32G0B1) on the Manta mainboard kept it current stat indefinitely! I tested it overnight >8h.

Expected behavior:
With a Linux host shutdown, the MCU should go into the default state (al off except hotend fan) like it works if "Firmware Restart" menu item (Mainsail UI) is pressed.

Sineos, 2022-09-27:
And as already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

FranzPVal, 2022-09-27:
Feature request???
No this is an issue. I asked on Discord The normal behavior was that the MCU goes into a reset / safe state if the connection to the host is lost. In my case it don't work this way.
If you tell me, that a printer mainboard keeps active (heater, fans, stepepr) if the Linux host is shutdown and this was the "normal" behavior, then klipper itself is not a safe working printer firmware and should not be used!
Please clarify the "normal behavior" for me.

FranzPVal, 2022-09-27:
some additional test results:
If one or both of the heaters active (hotend or bed) the desired behavior is present: The board goes into default state.
If none of the heaters are active, the board kepts it state (steppers, fans powered) forever.
---
I verified this with an test 2nd setup (BananaPi + Ender-3 Melzi style mainboard). So it's not a target specific bug but a general one.
I guess there is no timeout action for the case of host shutdown. the usual stepper timeout is handled by the host.

I agree that this is not a klipper bug. It's an odd and unwanted design flaw of klipper. So having a watchdog cmd and a mcu timeout handler is a feature request. I'm staggered that this is the case - as a developer I never ever expected this especially the host can't profit from the "frozen state" after reconnecting.

closed 2022-09-29 by github-actions[bot]