Klipper3d/klipper #5797: TEMPERATURE_WAIT macro blocks delayed gcode macros

vavaroutsos, 2022-09-27:
Can the delayed gcode scheduling be updated so that it is not blocked by execution of other gcode macros? I'm using delayed gcode for some custom temperature control, but the control macro is blocked from running (even though scheduled time has long past) by other gcode that take a long time to complete like TEMPERATURE_WAIT.

I would like to start my custom controls, then be able to wait until things stabilize by calling TEMPERATURE_WAIT. This doesn't work because TEMPERATURE_WAIT keeps the delayed gcode from running. I suspect it's the same for any gcode macro that happens to be running when delayed gcode is scheduled, but it's a real problem with TEMPERATURE_WAIT. Is it possible to have another thread to execute delayed gcode?

Sineos, 2022-09-28:
And as already stated when you opened the ticket: Issues here are for development only. Please be so kind to place your support / feature request on: https://klipper.discourse.group/

vavaroutsos, 2022-09-28:
Can you please clarify what qualifies for a github ticket? I consider this a bug, not a feature request. If I schedule delayed gcode to run in 5 seconds, but it doesn’t run for 20 minutes because something like TEMPERATURE_WAIT is running, this is a problem.

 

Best Regards,

 

Pete

vavaroutsos, 2022-09-28:
I’m still not understanding the requirements for creating a ticket on github. What exactly is meant by “development use”? Are you saying that only active developers should open github tickets and all user tickets, regardless of bug vs feature request, should be posted to discourse?

 

Best Regards,

 

Pete

closed 2022-09-30 by github-actions[bot]

Sineos, 2022-09-30:
Basically it is quite simple: 
1. Questions about Klipper
2. Feature requests
3. (Alleged) bugs
4. "Dissatisfaction" about current the current design (like this request)
5. Any other end user related topics

shall be posted on the discord / discourse. Your question can be categorized as either 2. or 4. so definitively "not on github".

The above aside: 
The core developers are also reading / following the above mentioned channels and the "collaborators", like myself, also bring certain topics to their attention actively.
You have absolutely no advantage / benefit in posting here instead of using the requested channels. 
Also the decision which topics / developments / requests are considered is ultimately their choice. Being a "hobby" that is provided free of charge to the audience and considering that the people have a private life as well this is equitable and should be respected.