Klipper3d/klipper #5813: special characters in exlude objects causing problems

Zodiak1993, 2022-10-07:
Hello, if you are using special characters for example "ö" in the name of your stl and create a gcode (with PrusaSlicer in my case) the print will not start. I am not sure if this is the expected behaviour. If this is not changeable maybe a warning would help users to find the problem. 
This happens only if you use [exclude_objects]

`UnicodeEncodeError: 'ascii' codec can't encode character u'\xfc' in position 50: ordinal not in range(128)`

[klippy(15).log](https://github.com/Klipper3d/klipper/files/9732786/klippy.15.log)

Sineos, 2022-10-07:
https://github.com/Klipper3d/klipper/issues/4794

closed 2022-10-07 by Sineos

Zodiak1993, 2022-10-07:
Thank you but in my case the filename is "testu.gcode" and the name of the object is "testü". I did not know that even the gcode file name itself has a problem with special characters.
But it is okay. I will be more careful when exporting files from fusion.