Klipper3d/klipper #5848: Cumulative error on Automatic Bed Mesh Y Axis

Behem07h, 2022-10-22:
Latest version of klipper, there seems to be an error on the bed mesh z offsets that accumulates along the Y axis

As shown in the images attached, SCREWS_TILT_CALIBRATE indicates that my bed is properly trammed. Yet vexingly, a 25x25 bed mesh returns a deviation of 0.6mm, but a 15x15 bed mesh returns a deviation of 0.15mm. I've also repeated for a 9x9 mesh and got a deviation of 0.1mm, and all 3 meshes had the same tilt upwards along the Y axis.

I'm not usually quick to blame things on a software issue. However, the fact that this error seemingly worsens the more points i probe along my Y axis makes me suspicious that it may be, since any physical cause would (should?) be location-consisent

These are the macros I am using, and I have tested with M190 S60 of CALIBRATE_BED_MESH_2 commented out to conduct the same tests at room temperature and ensure it is not a heat related issue with my bltouch

[gcode_macro CALIBRATE_BED_MESH_2]
description: Make a new bed mesh
gcode:
  M190 S60 #Wait for bed to reach 60C
  G28
  BED_MESH_CLEAR
  BED_MESH_CALIBRATE
  SAVE_CONFIG

[gcode_macro BED_AUTO_TRAM]
description: Use probe to calculate tramming distance
gcode:
  G28
  SCREWS_TILT_CALCULATE

[image]
[image]
[image]

Behem07h, 2022-10-22:
List of tests I've conducted, listed here for readbility sake:
1. Turned bed 90 degrees - same deviation along same axis
2. Checked x gantry - slightly twisted, but should not make a difference on a trammed bed
3. Checked bltouch mount - slightly twisted due to x gantry twist, but again should not make a significant difference on a trammed bed
4. Checked gantry rollers - using hiwin linear rails and no issues there
5. Tested with and without heat - error persists with and without bed being heated
6. Tried printing with the bed mesh anyway - high end of y axis is printing in mid air due to incorrect bed mesh

also: attached are my bed mesh settings
[image]

Behem07h, 2022-10-23:
no this is not a request for help

Behem07h, 2022-10-23:
could b0e related to #1394 #[https://github.com/Klipper3d/klipper/issues/5750](https://github.com/Klipper3d/klipper/issues/5750)

Behem07h, 2022-10-23:
even worse with a 75x75 mesh, look at this and tell me that this is somehow a hardware issue
if it really were, my hotend would have crashed into the bed by now - my z offset is only 1.8mm

[image]

jakep82, 2022-10-23:
Even if this is a bug in Klipper, we don't use GitHub to diagnose. If you want a response, please post on Discourse.

closed 2022-10-25 by github-actions[bot]