Klipper3d/klipper #5871: MPU9250/MPU6500 DEV_ID 0x70

tomas789, 2022-10-31:
I'm trying to attach the IMU to the Raspberry PI. I'm getting following error
```
$ ACCELEROMETER_QUERY
// Invalid mpu9250/mpu6050 id (got 70).
// This is generally indicative of connection problems
// (e.g. faulty wiring) or a faulty chip.
!! Invalid mpu9250/mpu6050 id (got 70).
```

After checking the [source code](https://github.com/Klipper3d/klipper/blob/24a1b50e512f3038a51060e3cc8ce2a847b9fafa/klippy/extras/mpu9250.py#L192) (link is to the exact version I'm running) I see that there are two hard-coded DEV_IDs.
```
MPU9250_DEV_ID =    0x73
MPU6050_DEV_ID =    0x68
```

But the problem is that I'm getting `0x70` from my chip. I can see that my device has a proper address
```
pi@fluidd:~/klipper $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

And I can also see that the value of the WHOAMI register (address `0x75`) is `0x70`
```
pi@fluidd:~/klipper $ i2cdump -y 1 0x68
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: c1 c8 df fe a8 34 13 e3 e9 ef 04 01 00 5e 59 7a    ?????4??????.^Yz
10: ba bc c0 00 00 00 00 00 00 00 00 00 00 00 00 00    ???.............
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
30: 00 00 00 00 00 00 00 00 00 00 01 23 0c 35 ec ff    ..........?#?5?.
40: 60 0b c0 01 e3 00 18 ff d5 00 00 00 00 00 00 00    `????.?.?.......
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
60: 00 00 00 00 00 00 00 00 00 00 00 01 00 00 06 72    ...........?..?r
70: 00 00 00 00 00 70 00 ea ec 00 15 22 00 25 88 00    .....p.??.?".%?.
80: c1 c8 df fe a8 34 13 e3 e9 ef 04 01 00 5e 59 7a    ?????4??????.^Yz
90: ba bc c0 00 00 00 00 00 00 00 00 00 00 00 00 00    ???.............
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
b0: 00 00 00 00 00 00 00 00 00 00 01 23 bc 36 20 fe    ..........?#?6 ?
c0: 40 0c f0 01 d2 00 23 ff c5 00 00 00 00 00 00 00    @????.#.?.......
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 01 00 00 09 00    ...........?..?.
f0: 00 00 00 00 00 70 00 ea ec 00 15 22 00 25 88 00    .....p.??.?".%?.                
```

This completely explains why I'm getting the exception. 

When consulting the [MPU6500 register map](https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6500-Register-Map2.pdf) I can see that the value `0x70` is as expected.

> 4.38 Register 117 – Who Am I
> Name: WHOAMI Serial IF: READ Reset value: 0x70
> 
> This register is used to verify the identity of the device. The contents of WHO_AM_I is an 8-bit device ID. The default value of the register is 0x70 for MPU-6500. This is different from the I2C address of the device as seen on the slave I2C controller by the applications processor. The I2C address of the MPU-6500 is 0x68 or 0x69 depending upon the value driven on AD0 pin.

As far as I understand it (and as [this document](https://www.cdiweb.com/datasheets/invensense/tdk-invensense-mpu9250-and-mpu6500.pdf) describes it) the MPU9250 is just two chips bundled into one. That is MPU6500 a 6DoF IMU and AKM-8963 3DoF magnetometer. The chip MPU6500 has address of `0x68` or `0x69` when accessed in the MPU9250. Problem is that I have a bare MPU6500 which has an address of `0x70`.

One other thing to add is that MPU9250 seems to be discontinued at the moment. This will probably mean that we will see more bare MPU6500 chips in the wild. Thus this might be relevant even more.

Based on my research, the `0x70` value should be added as a valid value too. Unfortunately I cannot experimentally verify this as there is some other problem with my setup preventing me from double-checking.

_Additional info_

[klippy.log](https://github.com/Klipper3d/klipper/files/9905502/klippy.log)

```
pi@fluidd:~/klipper $ git rev-parse HEAD
24a1b50e512f3038a51060e3cc8ce2a847b9fafa
pi@fluidd:~/klipper $ uname -a
Linux fluidd 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
```

Running on Raspberry PI 3B+.

KevinOConnor, 2022-11-01:
Sounds like you are looking for PR #5767.  If not, you'll probably want to open a new topic on Klipper Discourse, as we don't use github issues to investigate issues with a printer.

-Kevin

dingyifei, 2022-11-02:
yes, 0x70 is a valid id for mpu6500, and is included in the PR

closed 2022-11-19 by Sineos