Klipper3d/klipper #5875: Autogenerated Config Missing Keys

clairmont32, 2022-11-03:
Hi,

After running BED_MESH_CALIBRATE PROFILE=default METHOD=automatic I'm getting an error saying the autogenerated config doesnt have `mesh_min` specified. Similarly, I'm getting an error saying `control` is not present in the autogenerated config for the `heater_bed` section.

This is similar to [#3085](https://github.com/Klipper3d/klipper/issues/3085) based on searching with `is:issue autogenerated `

```
Option 'control' in section 'heater_bed' must be specified

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

Hardware:
Raspberry Pi 3B+
CR-10 Smart (Stock)
Config taken from a community post due to no template being available for this model. I'm willing to submit a PR for it after I get a decently working test print.

Latest log and config:
[klippy.log](https://github.com/Klipper3d/klipper/files/9932974/klippy.log)
[printer.txt](https://github.com/Klipper3d/klipper/files/9933018/printer.txt)

clairmont32, 2022-11-04:
it appears that `control: watermark` worked for me but having a duplicate bed mesh profile called "default" was confusing klipper. i removed the 2nd section of the mesh with the "default" name specified and it was happy again.

Arksine, 2022-11-04:
The original issue was the result of saving a profile with a null name.  The `[bed_mesh ]` section in there is invalid.   It appears that you added `mesh_min` and `mesh_max` to the default profile in an keep it happy, however those options don't belong in the autosave section.  FWIW, manually modifying the autosave section can result in unexpected behavior so I would recommend extreme caution when doing so.

It shouldn't be possible to save a profile to a null name, that would be a bug and something I'll look into.  In the meantime I would recommend removing anything related to `bed_mesh` from the autosave section.  When you run `BED_MESH_CALIBRATE` I would recommend omitting the `PROFILE` argument unless you want to save it to something other than `default`.

clairmont32, 2022-11-04:
Thanks for the prompt reply. I'll remove those entries and duplicate profile. I simply copy/pasted the command from the docs which has the profile name specified. Do you think it may be worth leaving the note of the default profile being there automatically but remove it from the code snippet?

clairmont32, 2022-12-11:
I think this can be closed since the fix was merged.

PR #5877 (2023-02-01): bed_mesh: profile fixes
This makes a couple minor changes to the profile behavior for `bed_mesh`:

1) Require that a profile name be specified for the `PROFILE` parameter in `BED_MESH_CALIBRATE` and all parameters in `BED_MESH_PROFILE`.  This resolves #5875.

2) Introduce an `initial_profile` option.  This allows users to specify which profile is loaded at startup, or to bypass loading a profile at startup by specifying a profile name that doesn't exist (my preference is to set this to none).
closed 2022-12-19 by JamesH1978