Klipper3d/klipper #5884: bl-touch on y-axis for a belt printer

JoshuaMorris, 2022-11-07:
Is it possible to add an option to define which axis the bltouch is configured for? I would like to setup my belt printer to use a bl-touch, but it looks like that is only possible for the z-axis. Are there any options I'm not seenig in the documentation?

Neko-vecter, 2022-11-09:
If my understanding currently. You have an infinite z? You wish to using y axis to probe the belt?

JoshuaMorris, 2022-11-09:
> If my understanding currently. You have an infinite z? You wish to using y axis to probe the belt?

 Exactly

Neko-vecter, 2022-11-09:
> > If my understanding currently. You have an infinite z? You wish to using y axis to probe the belt?
> 
> Exactly

Unfortunately the klipper currently does not really support infinite z. You probably need make a new kinematics to support this feature.

closed 2022-11-11 by github-actions[bot]