Klipper3d/klipper #5885: Unable to to mirror or swap A/B Motors in Hybrid CoreXY

liquid-light, 2022-11-07:
Hej

I would like to "open" this again because of the x+y and x-y topic.
I'm working on my setup since a few weeks now. But one thing I never solved is that this setup always prints mirroed parts.
Every Slicer assumes that xy 0 is front left. But with this setup you have basically no option to get this right (except for designing the whole printer setup according to the code).

With Corexy this is a simple fix. But not with the Hybrid Corexy (aka Markforged) kinematic.
First I thought I could use the mirror option for one axis (which is implemented in the IDEX) mode. But it seems that this is only available if you use a second carriage.

So basically I have two options right now. Either switch to Marlin again or changing the position of my Steppermotor.

_Originally posted by @liquid-light in https://github.com/Klipper3d/klipper/issues/4296#issuecomment-1147978886_

liquid-light, 2022-11-08:
Seriusly?
There is a feature in Klipper, that is in fact useless in the way it's designed. I point on this issue and then @Sineos comes around the corner and says "this is not an issue" and that's it?!? Interesting...

By the way...
Even the designer of this feature wrote that it needs to be reworked. I thought a seperate Issue that points to this topic (and is not closed like the other one) could help to track this. But obviously someone thinks Github is not designed to track issues...

jakep82, 2022-11-08:
What does opening a new issue accomplish?  You've already corresponded with the developer on the original PR multiple times. Tircrown has acknowledged it's an issue and when they have time they will work on it. If you need to use the printer now and Klipper won't work, then go back to Marlin.

liquid-light, 2022-11-08:
> What does opening a new issue accomplish? You've already corresponded with the developer on the original PR multiple times. Tircown has acknowledged it's an issue and when they have time they will work on it. If you need to use the printer now and Klipper won't work, then go back to Marlin.

Opening an issue helps other to see an issue. If someone have the time, knowledge and interest to help, he/she can help. If nobody knows that there is an issue, no one can help. It's that simple.
In addition an issue on Github helps to track the development.

Funny to explain this here on Github...

Obviously the original developer doesn't have time to work on the kinematics code. It's fine, Life happens. I know this very well and I really hope he and his family is OK! So maybe another experienced developer could solve this issue. But if nobody knows that it exists...

I would like to do it. But I'm not a software developer and far away from experienced enough. I'm mainly a hardware designer. A mechanical engineer. But luckily we have experienced people on many different fields. And if we work together we can make Klipper better.

You're asking what opening an issue accomplish. What does it accomplish if we do it this way?
"The feature doesn't work. Anyway, deal with it or use something else..."

closed 2022-11-10 by github-actions[bot]