Klipper3d/klipper #5886: Klipper not working

horsechick48, 2022-11-07:
Running a mini skr e3 v2 with bltouch, smart filament, dual extruder and Revo mini. 
I had this printer running ok but now all of a sudden it won't work. 
Can anyone go through this and tell me why it says bltouch cant verify it deployed and then it says end stop pin for it is not ok and then it won't move to the starting point. 

This is driving me nuts and I spent all day doing this. 

# This file contains common pin mappings for the BIGTREETECH SKR mini
# E3 v2.0. To use this config, the firmware should be compiled for the
# STM32F103 with a "28KiB bootloader" and USB communication. Also,
# select "Enable extra low-level configuration options" and configure
# "GPIO pins to set at micro-controller startup" to "!PA14".

# The "make flash" command does not work on the SKR mini E3. Instead,
# after running "make", copy the generated "out/klipper.bin" file to a
# file named "firmware.bin" on an SD card and then restart the SKR
# mini E3 with that SD card.

# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC0
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: PC11
tx_pin: PC10
uart_address: 0
run_current: 0.580
stealthchop_threshold: 999999

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC1
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_y]
uart_pin: PC11
tx_pin: PC10
uart_address: 2
run_current: 0.580
stealthchop_threshold: 999999

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0.0
position_max: 250

[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
run_current: 0.580
stealthchop_threshold: 999999

[extruder]
step_pin: PB3
dir_pin: PB4
enable_pin: !PD2
microsteps: 16
rotation_distance: 7.53
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
control: pid
pid_Kp: 35.054
pid_Ki: 4.972
pid_Kd: 61.782
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: PC11
tx_pin: PC10
uart_address: 3
run_current: 0.650
stealthchop_threshold: 999999

[heater_bed]
heater_pin: PC9
sensor_type: ATC Semitec 104GT-2
sensor_pin: PC3
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[heater_fan nozzle_cooling_fan]
pin: PC7

[fan]
pin: PC6

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_31FFD6053031543912780243-if00
restart_method: command

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[static_digital_output usb_pullup_enable]
pins: !PA14

[display] # RET6 12864 LCD
lcd_type: st7920
cs_pin: PB8
sclk_pin: PB9
sid_pin: PB15
encoder_pins: ^PA10, ^PA9
click_pin: ^!PA15

[safe_z_home]                         # enable for BLTouch
home_xy_position: 117.5,117.5
speed: 100
z_hop: 10
z_hop_speed: 5

[bltouch]                             # enable for BLTouch - fast-mode
sensor_pin: ^PC14
control_pin: PA1
pin_up_touch_mode_reports_triggered: True
pin_up_reports_not_triggered: True
stow_on_each_sample = False
probe_with_touch_mode: False
x_offset: -55                         
y_offset: -13                         
#z_offset: 1.6                        # modify as needed for bltouch or run PROBE_CALIBRATE
speed: 10
samples: 2
sample_retract_dist: 5.0              # Can be set lower, example 2.5 depending on height of bltouch from bed
lift_speed: 40
samples_tolerance_retries: 3
speed: 10

[bed_mesh]
speed: 250
horizontal_move_z: 5
mesh_min: 10, 10
mesh_max: 177.5, 217.5
probe_count: 5, 5

[filament_motion_sensor btt_smartie]
detection_length: 7.0
extruder: extruder
switch_pin: !PC15
runout_gcode: FILAMENT_RUNOUT

[gcode_macro FILAMENT_RUNOUT]
gcode:
    M600
[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
gcode:
    {% set X = printer.toolhead.axis_maximum.x|float - 5.0 %}
    {% set Y = printer.toolhead.axis_maximum.y|float - 5.0 %}
    {% set Z = params.Z|default(10)|int %}
    {% set E = params.E|default(1)|int %}
    SAVE_GCODE_STATE NAME=PAUSE_state
    BASE_PAUSE
    G91
    G1 E-{E} F2100
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F6000

[gcode_macro RESUME]
rename_existing: BASE_RESUME
gcode:
    {% set E = params.E|default(1)|int %}
    G91
    G1 E{E} F2100
    G90
    RESTORE_GCODE_STATE NAME=PAUSE_state MOVE=1
    BASE_RESUME

[pause_resume]
recover_velocity: 50.0

[gcode_macro M600]
gcode:
    {% set X = params.X|default(50)|float %}
    {% set Y = params.Y|default(0)|float %}
    {% set Z = params.Z|default(50)|float %}
    SAVE_GCODE_STATE NAME=M600_state
    PAUSE
    G91
    G1 E-.8 F2700
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F3000
    G91
    G1 E-50 F1000
    RESTORE_GCODE_STATE NAME=M600_state

[force_move]
enable_force_move: True

[virtual_sdcard]
path: ~/gcode_files

[temperature_sensor raspberry_pi]
sensor_type: temperature_host
min_temp: 10
max_temp: 100
     
     
[gcode_macro START_PRINT]
gcode:
    G28
        BED_MESH_CALIBRATE
        BED_MESH_PROFILE SAVE=p1
    G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
    G1 E50
    G1 X1 Y1 Z0.3 F5000.0 ; Move to start position
    M300 P250 ; play chime to indicate print starting
    G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
    G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
    G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
    G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
    G92 E0 ; Reset Extruder

[gcode_macro END_PRINT]
gcode:
           #Fix-up extruder
    G1 E-50
    #Present print
    G1 Z{printer.toolhead.position.z + 10} F600
    G1 X{X_MAX / 2} Y{Y_MAX} F6000
    M106 S0                                      
    #Disable Steppers
    M84 X Y E
    
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5,  EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8,  EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>
             

#[board_pins]
#aliases:
#  EXP1_1=PC6,EXP1_3=PB10,EXP1_5=PB14,EXP1_7=PB12,EXP1_9=<GND>,
#  EXP1_2=PB2,EXP1_4=PB11,EXP1_6=PB13,EXP1_8=PB15,EXP1_10=<5V>,
#  PROBE_IN=PB0,PROBE_OUT=PB1,FIL_RUNOUT=PC6
# See the sample-lcd.cfg file for definitions of common LCD displays.

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bltouch]
#*# z_offset = 1.6

Neko-vecter, 2022-11-09:
Are you using clone version BLtouch? 
If yes. This docs maybe help
https://www.klipper3d.org/BLTouch.html

straccio, 2022-11-10:
I have this problem too after the firmware update to v0.10.0-623-g5b1a6676 in a SKR 1.4

The probe stow correctly on power but no action on BLTOUCH_DEBUG

Sineos, 2022-11-10:
Please move this discussion to https://klipper.discourse.group as this topic is going to be closed anytime soon

closed 2022-11-12 by github-actions[bot]