Klipper3d/klipper #5901: Bug Connect Palette

3dinova, 2022-11-13:
I have a bug in my Klipper. I use Palette2 and it works fine. But now it updated and it just stopped working. The settings are fine, but when I give the command "PALETTE_CONNECT" an internal error occurs and Klipper shuts down.
[image]
[image]
[klippy.log](https://github.com/Klipper3d/klipper/files/9997427/klippy.log)

closed 2022-11-13 by Sineos

3dinova, 2022-11-13:
How is a bug reported to you? because even on discord they told me it was an update bug

Sineos, 2022-11-13:
See the bot's comment here: https://github.com/Klipper3d/klipper/issues/5900#issuecomment-1312663425

We do not use GH to diagnose potential bugs or support individual printer issues. The official place is the [Klipper Discourse](https://klipper.discourse.group)