Klipper3d/klipper #5958: Feature Request - Filament Runout Trigger 'PAUSE' After 700mm extruded  (distance_delay?)

phoenixfire425, 2022-12-28:
Would like the ability to set the filament runout trigger to be a set extruded distance (distance_delay?)

ie
when filament sensor triggers.  Wait 700mm before triggering PAUSE

manu7irl, 2022-12-28:
You can easily time the time it will take for the filament to go through
the seven hundred millimeters and then use the time flag before triggering
the pause

phoenixfire425, 2022-12-28:
> You can easily time the time it will take for the filament to go through the seven hundred millimeters and then use the time flag before triggering the pause
> […](#)

to many variables for that to be an accurate statement. 
If you are printing at 10mm/s vs 300mm/s and then accounting for nozzle size 0.2 != 1.0mm.  The flows can be dramatically different.

manu7irl, 2022-12-28:
Maybe a macro can be added to your start print and based on print flow and
speed you could alter the event_delay.

phoenixfire425, 2022-12-28:
> Maybe a macro can be added to your start print and based on print flow and speed you could alter the event_delay. On Wed, Dec 28, 2022, 8:31 PM phoenixfire425 ***@***.***> wrote:
> […](#)

I still think that this would be pretty inaccurate unless you parse the entire gcode file.  since slicers will slow down different section of the prints like walls, min cool time, infill, bridges, all are different speeds and flows.

even if you say.  flow rate is at 88%, and base speed is 150mm/s how would you determine the runout time delay?  specially not knowing how many variances are in the file.

for instance.  you have a test cube sliced at 300mm/s and a min cool down time of 10 seconds.  really you would only be printing around 20mm/s.

jakep82, 2022-12-28:
This is not the correct place for a feature request, nor a conversation about it. Please take this to Discord or Discourse.

closed 2022-12-31 by github-actions[bot]