Klipper3d/klipper #6066: Recent commit causes "OverflowError: can't convert negative number to unsigned" on mcu connect (STM32F407)

scottmudge, 2023-02-21:
Using a BTT GTR 1.0 board with an STM32F407.

I updated Klipper to node [d7bd7f1f4ba6cecd19daa566fdc1864561269ae1](https://github.com/Klipper3d/klipper/commit/d7bd7f1f4ba6cecd19daa566fdc1864561269ae1) yesterday, updated the board's firmware, and Klipper was no longer able to connect to the MCU over CAN:

```
mcu 'mcu': Starting CAN connect
Created a socket
Loaded MCU 'mcu' 119 commands (v0.11.0-112-gd7bd7f1f / gcc: (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027] binutils: (2.35.2-2+14+b2) 2.35.2)
MCU 'mcu' config: BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c3a=PH7,PH8 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_spi2b=PI2,PI3,PI1 BUS_PINS_i2c2a=PH4,PH5 BUS_PINS_spi2a=PC2,PC3,PB10 BUS_PINS_i2c1a=PB8,PB9 BUS_PINS_i2c3=PA8,PC9 RESERVE_PINS_crystal=PH0,PH1 RESERVE_PINS_USB=PA11,PA12 CLOCK_FREQ=168000000 BUS_PINS_spi3a=PC11,PC12,PC10 BUS_PINS_spi1a=PB4,PB5,PB3 STATS_SUMSQ_BASE=256 RECEIVE_WINDOW=192 BUS_PINS_sdio=PC12,PD2,PC8,PC9,PC10,PC11 RESERVE_PINS_CAN=PI9,PH13 STEPPER_BOTH_EDGE=1 CANBUS_BRIDGE=1 ADC_MAX=4095 BUS_PINS_spi3=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi1=PA6,PA7,PA5 PWM_MAX=255 MCU=stm32f407xx
mcu_temperature 'mcu' nominal base=-251.660377 slope=1236.226415
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/dietpi/klipper/klippy/klippy.py", line 180, in _connect
    cb()
  File "/home/dietpi/klipper/klippy/mcu.py", line 732, in _connect
    self._send_config(None)
  File "/home/dietpi/klipper/klippy/mcu.py", line 662, in _send_config
    cb()
  File "/home/dietpi/klipper/klippy/mcu.py", line 76, in _build_config
    state_tag), ffi_lib.free)
OverflowError: can't convert negative number to unsigned
webhooks client 140226830373904: New connection
webhooks client 140226830373904: Client info {'program': 'Moonraker', 'version': 'v0.7.1-885-g4550664'}
Got error -1 in can read: (100)Network is down
```

Reflashed just in case it was a compiler bug, same issue.

Reverted to node [aca0c71a2b2eb1679923ca2491aadad38cc43081](https://github.com/Klipper3d/klipper/commit/aca0c71a2b2eb1679923ca2491aadad38cc43081), recompiled and flashed, and it worked fine again.

Not sure if there is a bug in the SDIO code, but wanted to report the issue.

scottmudge, 2023-02-21:
There's no "Bug Report" section on the discourse forum. The closest applicable section is "General Discussion", and that entire section is 99% filled with newbie questions. Why are klipper related bugs not tracked here? That doesn't make any sense.

Sineos, 2023-02-21:
> This is clearly a bug with the commit I linked. Switching and recompiling between the two commits yields the issue, no other changes being made to the installation.

Since apparently this project does not show a link or anything else to create an issue, one could guess that it is not wanted to do so. Equally trying to outsmart this, one could also guess that this is not endorsed, wouldn't you agree?

scottmudge, 2023-02-21:
I suppose if one tries to click the button, but as someone who typically uses a keyboard more than a mouse, and who uses github frequently, I'm in the habit of manually typing in the URLs associated with issue reporting, PRs, etc. 

Which doesn't seem to redirect to the same diversion page the button appears to be tied to.

scottmudge, 2023-02-21:
Though I suppose that is a good method of eliminating a vast majority of the valueless "I can't connect, plz help" posts. A strategy I may employ for my own projects.

Sineos, 2023-02-21:
> Though I suppose that is a good method of eliminating a vast majority of the valueless "I can't connect, plz help" user-error related posts. A strategy I may employ for my own projects.

Exactly. And this is also the reason why we choose to triage all "bug" reports on discourse first. This eliminates like 99.987% of the "this must definitively be a bug, fix your software asap" reports (commonly known as PEBCAK).

Whereas I agree this looks like a legit report, please follow this philosophy nevertheless. The devs are reading on discourse as well.

scottmudge, 2023-02-21:
Sounds good, I can appreciate the approach taken. I'll stick to discourse/discord in the future.

Grafton1046, 2023-02-22:
I too ran into this error trying to get the USB can bridge working on my Ocotpus Pro
[klippy.log](https://github.com/Klipper3d/klipper/files/10800376/klippy.log)
Didn't need the CAN bus yet so reverted back to USB and everything is working now.

KevinOConnor, 2023-02-22:
I think this should be fixed now (commit d8811717).

As indicated earlier in this topic, in the future please report issues via Klipper Discourse.

-Kevin

scottmudge, 2023-02-22:
Confirmed fixed as of d881171 , thanks @KevinOConnor.

Closing

closed 2023-02-22 by scottmudge