Klipper3d/klipper #6196: Deadlock when using exclude_objects containing umlaut.

rufinus, 2023-05-09:
if you export from eg Fusion360 a multipart project as 3mf which uses umlaut in the naming of the objects. this names are used when slicing for the exclude_object_define block

```
 ; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
 ; 3 known objects
EXCLUDE_OBJECT_DEFINE NAME=Körper7_id_0_copy_0 CENTER=174.967,188.542 POLYGON=[[141.21,166.214],[141.21,205.789],[208.79,205
.789],[208.79,166.214]]
EXCLUDE_OBJECT_DEFINE NAME=Körper1_id_0_copy_1 CENTER=190.761,153.052 POLYGON=[[182.206,144.212],[182.206,161.787],[199.786,
161.787],[199.786,144.212]]
EXCLUDE_OBJECT_DEFINE NAME=Körper1_id_0_copy_0 CENTER=169.061,156.669 POLYGON=[[160.506,147.828],[160.506,165.404],[178.085,165.404],[178.085,147.828]]
```

If loaded in your interface of choice (in my case mainsail) the printer shows a pause state. unpausing fails with "not in pause state".
there is no warning/error displayed in the klippy.log or moonraker.log.

To reproduce you can use this 3mf - https://cloud.cwd.at/s/RYiyq7kZiSaTmCP

Sineos, 2023-05-09:
Known issue: https://github.com/Klipper3d/klipper/issues/4794

rufinus, 2023-05-09:
Its not a problem with the filename - its a problem with the object name!

iqubik, 2023-07-27:
No solutiuons?