Klipper3d/klipper #6228: MCU Problem

GiacomoC00, 2023-05-27:
I have klipper installed in my RPi4 and I am using it with my Sidewinder X2. Today I updated the OS packages and Klipper to the version released the 25th of May and after that I was not able to retrieve the MCU, even flashing again the new built firmware. The only way to solve was to reinstall Raspberry OS (but this time without doing the OS packages update through Mainsail)and Klipper 1 commit back. Am I the only one that had this problem? Thank you.

closed 2023-05-27 by Sineos

lnorton89, 2023-05-28:
I'm also experiencing this issue after update. 
[image]

GiacomoC00, 2023-05-28:
> I'm also experiencing this issue after update. [image]

Hi, I solved doing this: https://klipper.discourse.group/t/debian-bullseye-bug-causing-klipper-to-no-longer-find-the-printer-board/8231

lnorton89, 2023-05-28:
Confirmed this fixes, thanks buddy.