Klipper3d/klipper #6338: Apparent regression in variable PA

richfelker, 2023-09-08:
Commit 29724a741106 moved the PA constant from being a property of the move to a property of the extruder. As far as I can tell, this broke configurations with variable PA, e.g. use of a lower K for bridges where there's no backpressure, by performing the integration with a single K that's whatever happens to have been the most-recent-set at the time of the iterative solver running, rather than using potentially differing-per-move K values in the integration.

My proposal for fixing this would be reverting the changes to chelper and the representation of PA K in trapq, and instead having the Python side copy the PA K from the relevant extruder object to the moves when adding them to the trapq.

richfelker, 2023-09-09:
This is not a request for help, silly bot. It's a report citing a particular commit and describing the regression it introduced.

Sineos, 2023-09-09:
As you surely noticed, this project has no possibility to open a new issue. To simply outsmart this fact seems rather brazen to me.
Please either open a PR or follow the given instructions.

closed 2023-09-09 by Sineos

ganzgustav22, 2024-01-26:
@Sineos But apart from richfelker being a naughty boy, have you maybe considered fixing it?

Sineos, 2024-01-27:
You might want to open a topic in Discourse as this is still not considered the right place to discuss it.

ganzgustav22, 2024-02-01:
No thanks, I'm happy with Danger Klipper.

richfelker, 2024-06-24:
Opened thread in the proper place here: https://klipper.discourse.group/t/regression-in-dynamic-pa/17260