Klipper3d/klipper #6421: Klipper fails to build with Apple's cpp

dottedmag, 2023-12-09:
I'm building STM32 target for Klipper on MacOS 13.6. It fails with a cryptic message:

```
$ make
...
  Preprocessing out/src/generic/armcm_link.ld
clang: error: no input files
make: *** [out/src/generic/armcm_link.ld] Error 1
$
```
Tracing the build shows the following:
```
$ make V=1
...
cpp -Iout/ -P -MD -MT out/src/generic/armcm_link.ld src/generic/armcm_link.lds.S -o out/src/generic/armcm_link.ld
clang: error: no input files
make: *** [out/src/generic/armcm_link.ld] Error 1
$
```
And indeed running this command fails:
```
$ cpp -Iout/ -P -MD -MT out/src/generic/armcm_link.ld src/generic/armcm_link.lds.S -o out/src/generic/armcm_link.ld
clang: error: no input files
$
```
This preprocessor is from Xcode:
```
$ cpp --version
Apple clang version 15.0.0 (clang-1500.0.40.1)
Target: arm64-apple-darwin22.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
$
```

The preprocessor coming with GCC works:
```
$ make CPP=arm-none-eabi-cpp
...
  Creating hex file out/klipper.bin
$
```

`CPP` is set to `cpp` in Makefile, I guess changing it to `$(CROSS_PREFIX)cpp` is not going to break anything.

## Environment

- MacOS 13.6 (arm64)
- gcc is from Homebrew `gcc-arm-embedded`
- Native toolchain is from Xcode 15.0.1

dottedmag, 2024-01-13:
bugs don't fix themselves.

gooodgy, 2024-01-17:
Have you tried rolling back to version 10.3 of gcc? This fixed this problem for me.

dottedmag, 2024-01-17:
This change has nothing to do with gcc. It is about _selecting_ GCC's preprocessor instead of Apple-provided LLVM-derived preprocessor.

gooodgy, 2024-01-17:
Well did your earlier fix do the trick? I found that downgrading to gcc 10.3 combined with your fix enabled me to solve this issue.

gooodgy, 2024-01-17:
So... are you still having problems or not?

gooodgy, 2024-01-17:
Well that's good. Just for reference, what version of gcc do you have?

dottedmag, 2024-01-25:
@sxming1010 I guess you've got the wrong issue to post this comment to.

sxming1010, 2024-01-25:
I tried to use the MPU6050 gyroscope combined with the MDXL345 to realize the exploration of controlling the fan to always be perpendicular to the tangent direction of the nozzle motion curve trajectory. Whether it is necessary to completely change the jitter compensation algorithm, I have to re-explore and edit it myself.

dottedmag, 2024-02-29:
Bugs don't fix themselves

wildblue, 2024-03-03:
Hello,
i have noticed that a comment like the following one makes Klipper unable to read the file or start up fully anymore:
# Temp  above 25°C

##me bitching about my terrible life##
I tried to notify about this in the discord App, installed it just for this case. I wanted to help this great project.

First I was told this was a user error, i was informed my commenst had been moved and realized they have been deleted. To my wondering i was blocked for an hour because i have NOT read everything in two different topics before it was displayed on my screen (laggy Internet...).

Sorry for commenting here, but i still think it is NOT an user error when using characters printed on a fully standard german keyboard layout.

I want to be able to have a discussion about that without being told i could write a  PR (as end-user asking for help) to fix this because it was very unlikely to be fixed anyways. 

I am fully aware of my misuse of this thread in your bugtracker and promised that  i will never again be bitching like just now in your github publicly. I am deeply sorry for not being able to Just let Go with this.

Sineos, 2024-03-03:
> ##me bitching about my terrible life##

I'm not sure if I understand your issue. However, I am certain that this is not the appropriate place or method to address your frustration. 
Please create a post on https://klipper.discourse.group, and we will work to resolve the issue. Thank you.

dottedmag, 2024-04-08:
bugs don't fix themselves

KevinOConnor, 2024-04-09:
The Klipper github issue tracker is no longer used for any purpose.  None of the active developers are reading or tracking any of the messages here.  If there is an issue or something to discuss with the developers see the directions at: https://www.klipper3d.org/Contact.html

-Kevin

closed 2024-04-09 by KevinOConnor