Klipper3d/klipper #6619: Clock sync and unrealistic extruder temps after restart

chuckwagoncomputing, 2024-06-15:
I have been Mildly Annoyed, and decided to do something about it.

I have been having the same issue as some other folks:
[@viesturz ](https://klipper.discourse.group/t/ebb36-temp-sensor-spikes-on-klipper-restart/11601)
[Cosmin](https://klipper.discourse.group/t/reporting-wrong-extruder-temp-after-restart/11348)
[Not sure if the same issue, but seems likely](https://klipper.discourse.group/t/erroneous-rtd-amplifier-temperatures-during-amplifier-initialization-when-save-config/15022)

In short, when using a SB2209 CAN bus toolhead board, when restarting Klipper with `RESTART`, the extruder temperature is reported as a very large number, either negative or positive, before returning to normal over the next ~10 seconds. This doesn't happen all the time; it's pretty random.

Here is the cause as well as I can understand it:

1. The toolhead keeps sending `thermocouple_result` commands, because it wasn't restarted.
2. When a `thermocouple_result` command is sent before clock sync finishes, `SecondarySync`'s [`clock_adj`](https://github.com/Klipper3d/klipper/blob/fcf064ba6851042a12a71975c229a9670b34cf31/klippy/clocksync.py#L167) still has 1 for the mcu frequency, which results in spi_temperature.py's [`_handle_temperature_response`](https://github.com/Klipper3d/klipper/blob/fcf064ba6851042a12a71975c229a9670b34cf31/klippy/extras/spi_temperature.py#L55) getting an undivided value from its call to [`clock_to_print_time()`](https://github.com/Klipper3d/klipper/blob/fcf064ba6851042a12a71975c229a9670b34cf31/klippy/clocksync.py#L184).
3. If the temperature reading has changed on the first `thermocouple_result` command after clock sync has finished, [`temperature_callback`](https://github.com/Klipper3d/klipper/blob/fcf064ba6851042a12a71975c229a9670b34cf31/klippy/extras/heaters.py#L82) in heaters.py will multiply the difference in temperature by the difference between the undivided value and the new correct value, resulting in a very large "smoothed" temperature, either negative or positive depending on the direction of the temperature change.

On my machine, I implemented the following fix:

In [`_handle_spi_response`](https://github.com/Klipper3d/klipper/blob/fcf064ba6851042a12a71975c229a9670b34cf31/klippy/extras/spi_temperature.py#L55) in spi_temperature.py:
```diff py
        last_read_time  = self.mcu.clock_to_print_time(last_read_clock)
+        if last_read_time == last_read_clock:
+            logging.warning("Ignoring untrustworthy clock sync")
+            return
        self._callback(last_read_time, temp)
```

I'm not sure if this is the appropriate fix, as this is the first time I've touched the Klipper codebase, but I rather doubt it. It feels like something more fundamental should change, like maybe `clock_to_print_time` should throw an error if the clock isn't synced.

Sineos, 2024-06-15:
We do not use GH to discuss such topics. Please refer to above link.

closed 2024-06-15 by Sineos

chuckwagoncomputing, 2024-06-15:
Did you read it? This is not a request for help, a feature request, a question, or a request for diagnosis.
I found a firmware bug, and presented a solution for comment. Would you rather I open a pull request with my half-baked fix?

Sineos, 2024-06-15:
https://www.klipper3d.org/Contact.html#i-found-a-bug-in-the-klipper-software