Klipper3d/klipper #6693: Use of negative error code

Laikulo, 2024-09-19:
Negative error codes are reserved by convention for when a process is killed by a signal N, where the RC is -N.

This behavior is apparently a holdover from actual UNIX, so strong docs are hard to find. The following is form the POSIX manpange for exit

```
    [...] certain exit status values have been reserved for special uses and should be used by applications only for those purposes:

     126
        A file to be executed was found, but it was not an executable utility.
     127
        A utility to be executed was not found.
    >128
        A command was interrupted by a signal.

```

Note that the exit value is 8-bit unsigned or masked to 8 bits (on linux), so -1 complements to 255, unless undefined behavior (below) is occurring.

Python docs also call out that values outside of 0-127 have undefined behavior: https://docs.python.org/2/library/sys.html#sys.exit

https://github.com/Klipper3d/klipper/blob/87ac69363aecd51c282999f8bf0f5c3513f58ed2/klippy/klippy.py#L364

bevanweiss, 2024-09-19:
Is your proposal that a value of '1' would be returned instead?

Laikulo, 2024-09-19:
> Is your proposal that a value of '1' would be returned instead?

That, or any other value in 1-125.

Laikulo, 2024-09-19:
I'll migrate this to a discourse thread later today. Not quite sure where it's supposed to go in there.

bevanweiss, 2024-09-19:
> > Is your proposal that a value of '1' would be returned instead?
> 
> That, or any other value in 1-125.

You'd probably be better off just raising a Pull Request with your proposed 'fix'.

Sineos, 2024-09-19:
> Not quite sure where it's supposed to go in there.

This should go into the Developer section. 
It would probably be useful to discuss IF and HOW such exit codes could be managed, and in particular, how this would ultimately benefit the real world.

closed 2024-09-21 by github-actions[bot]