Klipper3d/klipper #6728: "GPIO pins to set at micro-controller startup" possibly not working

VinnyCordeiro, 2024-11-02:
Decided to do a new Klipper installation on my printer, which was still running version 0.10, so did a clean Raspberry Pi OS install on a separate microSD card, proceeded to install Klipper/Moonraker/Mainsail on it, copied the relevant config files just modifying the sections that were not needed/deprecated, and updated the firmware on the board, a BTT mini E3 V2.0, with the proper `GPIO pins to set at micro-controller startup` set with `!PA14` as instructed on the sample config file.

After rebooting the board was not connected to the Raspberry Pi 4 via USB, confirmed via `ls /dev/serial/by-id/*`, and since this was a reinstall I knew the USB cable was good, which means the `!PA14` was not working (it needs to be that way in order to enable the USB port on this particular board).

I was only able to make the board work again by turning on the option `Disable SWD at startup (for GigaDevice stm32f103 clones)`. Please check this.

closed 2024-11-04 by github-actions[bot]

VinnyCordeiro, 2024-11-05:
Jesus Christ, this is a possible BUG REPORT, not an ask for help. I was able to solve the problem using a not documented procedure, why would this be classified as a request for help?

Unfortunately this is the kind of situation that is driving people towards Klipper forks that are maintained by developers who actually listen to the users. I hope that change in the future, but I won't hold my breath.

jakep82, 2024-11-08:
> Jesus Christ, this is a possible BUG REPORT, not an ask for help. I was able to solve the problem using a not documented procedure, why would this be classified as a request for help? [...]

If you click on the contact link and read what it says, you'll see that we request all issues be posted on Discourse. Developers do read and respond on Discourse where appropriate. They generally don't respond to any issue opened here. 

For reference the only reason the issue tracker on GitHub hasn't been entirely disabled is so past issues remain visible and searchable.