Klipper3d/klipper #6822: Hiding past errors or what?

KimmoHop, 2025-02-14:
Other open source projects in social media: _did you create issue in github?_
Klipper github: _did you create issue in social media?_

I find it disturbing to wipe part of project history (4k+ issues resolved!) and hide _/ move / choose you term here_ new issues and improvement ideas from main repository to 3rd party platform. Can you explain the reasoning for this and who made such decision? Is klipper still open and welcome to people not belonging to core group?

I know that e.g. Cura and Prusa have their internal repositories and issue management systems, but they still have open github access to community users and developers. Everyone benefits from that.

Sineos, 2025-02-14:
The decision is simple:
* Development history is fully transparent via the respective PRs and commits.
* "Legacy" issues are still available and will continue to be so for future reference.
* The GH issues are not fully blocked or turned off because it would kill all old issues. 
* New issues are no longer handled via GH, as the history and experience have shown that more than 99% are not bugs but support requests.
* For the details please refer to https://www.klipper3d.org/Contact.html

closed 2025-02-16 by github-actions[bot]

mattswest, 2025-06-24:
>     * New issues are no longer handled via GH, as the history and experience have shown that more than 99% are not bugs but support requests.

This project is run with the mentality of anything that **can** be blamed on the user **should** be blamed on the user, which is pretty absurd considering:

1. The documentation is scattered and incomplete. Just look at how GCode commands are documented on the Marlin website compared to how Klipper commands are documented. So how are users supposed to figure how what is correct anyway?
2. An abysmal lack of any sensible logging or error output. I guess timestamps are just too much to ask for? IDK, maybe this has to something to do with sensitive timing requirements.
3. Some nonsensical or unjustifiable defaults. You create situations where anyone who tries to use a specific feature runs into an issue, and then get upset when all of those users come looking for help.

The bottom line is, this project as a whole completely disregards the value of user feedback. It sees no potential value in smoothing over potential rough spots in the user experience, even if resulting from shortcomings of the software itself. All that matters is that the software functions with technical proficiency if configured correctly.

KimmoHop, 2025-06-24:
@mattswest My choice: Marlin + Octoprint. I get better printer UI with DWIN than with mobile phone (app or browser) and can get or develop plugins to smoother host-client interface.

It's **good** that Klipper is **open source**, but with more printers having "closed" Klipper and dedicated user forums and support systems, difference to true closed source firmware dilutes. Unless targeting for _INDX_. My next printer selection will be based mainly on print quality and nozzle availability.

biorpg, 2025-10-07:
> This project is run with the mentality of anything that **can** be blamed on the user **should** be blamed on the user, which is pretty absurd considering:
>
>     1. The documentation is scattered and [...]

Your points 1, 2, and 3 are valid, but your preface and 'bottom line' are a bit of a stretch. In any case, while I agree that the move is kind of an anti-social thing to see from an open-source project, I don't think it is unreasonable or unacceptable. They are just deferring to a pretty standard strategy to filter out noisy nuisances that eat up the time they have available to work for free on a project that you and others find useful- adding an additional layer for the majority of users to traverse, should they have a meaningful issue to report. It can seem abrasive, but consider how irritating the small scuff that prompted you to share these thoughts really hurt to the  possible mind-numbing droll of answering the same non-issues or already documented questions or questions that should really be for other projects to answer.